gsilano/CrazyS #96: Is it possible to forcefully stop the drone/s when it reached a target location?

rglbiv, 2022-07-13:
I am using the [Interfacing CrazyS through MATLAB](https://github.com/gsilano/CrazyS/wiki/Interfacing-CrazyS-through-MATLAB) as the base of my project and I already have a controller that can move a drone team in six directions (forward, backward, left, right, up, down). 

I am exploring a new function in my controller that will let the drone go to a desired target location _x_,_y_. But I am having a issue similar to #1 where "when I publish a waypoint the quadrotor does not stop at the goal pose and continues to fly in the direction it had to arrive the goal". Afterwards, the drone will self-correct itself, and move back to its target location _x_,_y_.

This [video](https://youtu.be/JR_2cW6uQJk) demonstrates the issue. It's still in development/debugging stage, but the goal of the drones here is to arrive at the location wherein the ball is at the center of the ring.

My questions are:
1. Is it possible to forcefully stop the drone/s when it reached a target location?
2. If the answer to question 1 is a yes, which parameters seen in the image below should I tinker with? (all my codes are done within Simulink)
[image]

Thank you for taking your time in reading and hopefully answer my query!

gsilano, 2022-07-14:
Hi @rglbiv! First a very nice video. The work seems to be very interesting. Unfortunately, the current control algorithm is not meant to be used as a waypoint controller. You need to design your own controller or you may follow the direction of [this thread](https://github.com/gsilano/CrazyS/discussions/94). I've started working on it, but there is still something to be done.

gsilano, 2022-07-16:
The controller continues to use the previous value published on the ROS topic until the next input is provided. I hope this helps.

rglbiv, 2022-07-16:
I see. I’m glad my understanding is correct. I will continue exploring options to arrive at the answer I am looking for. Thank you as always.

closed 2022-07-16 by rglbiv