gsilano/CrazyS #100: opticalFlow.pb.h:

omr-web, 2022-12-02:
I want to use only opticalflow for my drone simulation. I get opticalflow plugin and I added to sdf as sensor plugin but when I compile the code, I see this error  "fatal error: opticalFlow.pb.h: No such file or directory". Could you help me please?

gsilano, 2022-12-05:
Bye! Where did you get it? Could you be more specific? Below are the links to two files where the optical flow is included in the Gazebo model: [LINK1](https://github.com/ethz-asl/rotors_simulator/blob/7f6b5529bfc793e77275ca32d8e4970a0c755e5d/rotors_gazebo/models/flow_cam/model.config) and [LINK2](https://github.com/ethz-asl/rotors_simulator/blob/7f6b5529bfc793e77275ca32d8e4970a0c755e5d/rotors_gazebo/models/flow_cam/model.sdf).
You could use them to check if you included it correctly or not.

This could be the case with a similar number of #40. I mean, you have to copy and paste a missing object file.

gsilano, 2026-10-13:
Closed for inactivity

closed 2026-10-13 by gsilano