gsilano/CrazyS #101: Question regarding posting motor speed to Gazebo & some theoretical parameters

wonggwan, 2023-01-27:
Hi,

Thanks for the great repo! As I sent the email a few days ago that probably was not clear enough to state my question, I am sending all the related issues I have using this repo over here:)

## My setup

I am using **Ubuntu 20.04 with ROS Noetic and Gazebo 11 version** and have already managed to setup & install all the things now

## Issue 1: publishing motor speed to Gazebo via rostopic

After I managed to show the CrazyS robot in Gazebo, I was trying to control the drone via Python script by publishing the corresponding motor speed to the ROS topic. On my side, when I type `rostopic list`, the following topics were shown:

```
/clock
/crazyflie2/command/motor_speed
/crazyflie2/command/roll_pitch_yawrate_thrust
/crazyflie2/gazebo/command/motor_speed
/crazyflie2/ground_truth/imu
/crazyflie2/ground_truth/odometry
/crazyflie2/ground_truth/pose
/crazyflie2/ground_truth/pose_with_covariance
/crazyflie2/ground_truth/position
/crazyflie2/ground_truth/transform
/crazyflie2/joint_states
/crazyflie2/joy
/crazyflie2/joy/set_feedback
/crazyflie2/motor_speed
/crazyflie2/motor_speed/0
/crazyflie2/motor_speed/1
/crazyflie2/motor_speed/2
/crazyflie2/motor_speed/3
/crazyflie2/odometry
/crazyflie2/odometry_sensor1/odometry
/crazyflie2/odometry_sensor1/pose
/crazyflie2/odometry_sensor1/pose_with_covariance
/crazyflie2/odometry_sensor1/position
/crazyflie2/odometry_sensor1/transform
/crazyflie2/orientation_rpy
/crazyflie2/pose
/crazyflie2/pose_with_covariance
/crazyflie2/position
/crazyflie2/transform
/crazyflie2/wind_speed
/diagnostics
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/performance_metrics
/gazebo/set_link_state
/gazebo/set_model_state
/rosout
/rosout_agg
/tf
/tf_static
```

**_My questions are as follows for the first issue:_**

- As there seems to be multiple topic that is related to motor speed, **which is the one that I should publish my information to and what is the unit of the motor_speed we publish (rpm or rad/s)?** 
```
/crazyflie2/command/motor_speed
/crazyflie2/gazebo/command/motor_speed
/crazyflie2/motor_speed
/crazyflie2/motor_speed/0
/crazyflie2/motor_speed/1
/crazyflie2/motor_speed/2
/crazyflie2/motor_speed/3
```

- I tried to publish my motor speed to the topic `/crazyflie2/command/motor_speed` which is using the message type `mav_msgs/Actuators`, but when I find the corresponding information about this message type in [ROS webpage](http://docs.ros.org/en/indigo/api/mav_msgs/html/msg/Actuators.html), it shows

```
float64[] angular_velocities # Angular velocities of the actuator in [rad/s].
                             # E.g. "rpm" of rotors, propellers, thrusters 
```
But I think the motor speed has a different unit `rpm` compared to the angular velocity of the actuator `rad/s`, **how do we specify this when we publish our motor speed to Gazebo**?

- What is the **correct max angular velocity** of the CrazyS in Gazebo? Is it 2618 rad/s?

## Issue 2: question regarding some coefficient & parameter setting to calculate the system dynamics of the CrazyS

**_My questions are as follows for the second issue:_**

- As I found in [your publication](https://giuseppesilano.net/publications/rosChapter19.pdf) on page 6:

[image]

**Does the parameter match the physical configuration of the drone being used in Gazebo?** 

- What is the **correct MASS of the drone as a whole** (including the motors)? I checked in Gazebo it showed as 0.03 kg, yet the `CrazyS/rotors_description/urdf/crazyflie2.xarco` showed that:

```
  <xacro:property name="mass" value="0.025" /> <!-- [kg] -->
  <xacro:property name="mass_rotor" value="0.0005" /> <!-- [kg] -->
```

which means the total mass will be 0.025 + 4*0.0005 = 0.027 kg.

Thank you so much!

rglbiv, 2023-01-29:
Hi! Think of my 'answers' with a grain of salt, I am no expert regarding this repo. I am just someone who used this repo recently as well.

For Issue 1, you publish the motor speed to `/crazyflie/command/motor_speed`. I used the Matlab version of this repo, so I am not entirely sure how to publish
via script, but in Matlab, `/crazyflie/command/motor_speed` needs input from `mav_msgs/Actuators` and Propellers Angular Velocities (4) from Omegas (motor speed), see Fig. 11 (p.22) of the [documentation](https://giuseppesilano.net/publications/rosChapter19.pdf) that you refer to. 
As for the units, I already forgot but I think it's in the docu as well.

As for Issue 2, I think it's correct as maybe the ones showed in Gazebo is rounded-up to 0.XX significant digits? Not entirely sure.

If you wait for a bit, your issues maybe resolved by the author of this repo, good luck!

wonggwan, 2023-01-29:
Update for Issue 1 regarding the motor speed:

I found that the `/CrazyS/src/CrazyFlie/rotors_control/src/nodes/roll_pitch_yawrate_thrust_crazyflie_node.cpp` is also publishing some sort of motor speed commands to gazebo apart from the `/crazyflie2/command/motor_speed` one, what should we do about this? E.g. does these two node interfere each other during simulation in gazebo?

wonggwan, 2023-01-29:
> Hi! Think of my 'answers' with a grain of salt, I am no expert regarding this repo. I am just someone who used this repo recently as well.
>
> For Issue 1, you publish the motor speed to [...]

Hi! Thanks for the help! 

For Issue 1, now I also agree with you that we should publish the motor speed to the `/crazyflie2/command/motor_speed`, but as my comments above mentioned, there is one more node named as `/CrazyS/src/CrazyFlie/rotors_control/src/nodes/roll_pitch_yawrate_thrust_crazyflie_node.cpp` that is also publishing something towards the same topic, so I am not sure how is this affect my result.

As for Issue 2, since the drone is too light-weighted (roughly 30 grams), so any small change in the value will result in a very huge change in the calculation result. Since I am trying to use pure math calculation to 'simulate' how the drone fly, so I think the numbers will be crucial to the result.

gsilano, 2023-01-31:
Hi wonggwan! Glad to hear from you that you found my work helpful. Here are my answers:

* Motor speeds are expressed in radians per second [rad/s]. If you want to control the drone, without using the provided controller, just run `roslaunch rotors_gazebo crazyflie2_without_controller.launch`. Then, your code should publish on the `/crazyflie2/gazebo/command/motor_speed` topic.

* ROS Messages are defined in the [`mav_msgs` repository](https://github.com/gsilano/mav_comm). Specifically, here you can find the definition of [Actuators](https://github.com/gsilano/mav_comm/blob/master/mav_msgs/msg/Actuators.msg). You can find the maximum angular velocity in the the `crazyflie2_base.xacro` and `crazyflie2_forster.xacro` files. Since I have never tested the simulator with the Crazyflie, comparing the behaviors, I suggest you find your own parameters or rely on those in [Forster's thesis](https://www.research-collection.ethz.ch/handle/20.500.11850/214143).

I hope this helps. If not, let me know.

wonggwan, 2023-01-31:
> Hi wonggwan! Glad to hear from you that you found my work helpful. Here are my answers:
>
> * Motor speeds are expressed in radians per second [rad/s]. If you want to control the drone, without using [...]

Thank you so much for your help! I think that solves my current problem! Closing this issue now:)

closed 2023-01-31 by wonggwan