gsilano/CrazyS #103: Unrecognized message type mav_msgs/Actuators.

nrlhozkan, 2023-07-16:
Hi, I have a problem with message type mav_msgs/Actuators. I created the message and when I do rosmsg list, I can see it in the list. But when I wanto create a message with selecting this type matlab says the 'Unrecognized message type mav_msgs/Actuators. Use rosmsg list to see available types'. How did you solve this problem? Thank you

nrlhozkan, 2023-07-18:
Yes, I have read it. I have tried to do it with Matlab script and Simulink. But it same error "Unrecognized message type mav_msgs/Actuators".

gsilano, 2023-07-31:
I hope this [guideline](https://it.mathworks.com/help/ros/ug/ros-custom-message-support.html) helps

nrlhozkan, 2023-08-01:
Thank you but I tried that one too but still the same result. I can send messages using Python but MATLAB/Simulink does not allow it. I do not know why. I also have another problem maybe you would know it. This time in Python. I have my own controller for crazfiles based on feedback linearization.  I am using ode to integrate my differential equations. In each iteration, I would like to publish the actuator speeds. However, I am facing issues probably related to time rates. Have you ever faced such an issue?

gsilano, 2023-08-03:
Hey there! I've been through this issue before. It seems like it affected earlier releases of the Robotics System Toolbox. I don't use it anymore; instead, I use [Genom](https://www.openrobots.org/wiki/genom) for my MATLAB/Simulink simulations. However, this is another thread not related to this issue. Try out this guideline I wrote: [LINK](https://github.com/gsilano/CrazyS/wiki/Interfacing-CrazyS-through-MATLAB).

Back then, I found a simple workaround using ROS services to freeze and resume Gazebo simulations and Simulink for message reading. It was slow, but it allowed me to control the quadrotor. You can find the [MATLAB script here](https://github.com/gsilano/CrazyS/wiki/files/RoboticsSystemToolboxExampleFile.rar), along with additional scripts on that page.

As for the rate issue, I honestly can't recall how I fixed it. It's been quite some time. However, you could take a look at the code to see if it helps. If you're okay with Italian (sorry, I haven't had time to translate this contribution), there's a useful repo [here](https://github.com/sgrossi/estensione_crazys).

By the way, Stefano managed to come up with a simple script for controlling the Crazyflie using MATLAB and ROS messages. You can find it in the `matlab_lee_controller` folder. An example of this script can be found [here])(https://github.com/gsilano/CrazyS/wiki/files/matlab_lee_controller.zip)

nrlhozkan, 2023-08-08:
Thank you for your reply. I will let you know if they help in future.

closed 2023-08-08 by nrlhozkan