gsilano/CrazyS #104: Noetic Controller Package

furkan-cetin1453, 2023-07-23:
I am using the Noetic distribution. Is there a controller package written for this distribution? Or is there a tutorial you can recommend on how to write my own controller package? I looked at the problems in general. But I am confused about this. I want to work on swarm formation structures. For this, I need to have knowledge of how the vehicle moves in 4 directions and how it rotates around itself. I want to do this through topics

gsilano, 2023-07-24:
Hi! I didn't get what you are looking for. Are you looking for an existing controller package compatible with the Noetic distribution to control the Crazyflie, or are you interested in creating your own custom controller package?
If you're referring to creating your own algorithm, consider looking at [reference publications](https://github.com/gsilano/CrazyS/wiki/Publications) for insights.

furkan-cetin1453, 2023-07-25:
Yes. I am looking for a package compatible with the Noetic distribution.

gsilano, 2023-07-31:
I see. I can suggest taking a look at the following ROS packages: [Crazyswarm](https://github.com/USC-ACTLab/crazyswarm), [Crazyflie_nmpc](https://github.com/bcbarbara/crazyflie_nmpc), and [Crazychoir](https://github.com/OPT4SMART/crazychoir). I hope this helps.

closed 2023-10-13 by furkan-cetin1453