gsilano/CrazyS #105: rotors_gazebo_plugins:make error

NoemiF99, 2023-09-22:
[image]
[image]
[image]
How can I solve this problem?

gsilano, 2023-09-24:
Hello! It's a bit unclear to me what's happening in that context. Could you please provide more information regarding the Ubuntu and ROS releases you're referring to? Thank you!

NoemiF99, 2023-09-24:
hello, I have ubunto 20.04.2.0, ros noetic e Gazebo multi-robot simulator, version 11.11.0 with Virtual Box. Installed CrazyS following ReadMe.md and when i run 'catkin build', I encounter the issues i shared in my previous post. I read solution #590, which had the same error as mine, and it said to follow the ReadMe.md, which I followed to use CrazyS, but every time I try to 'catkin build,' it gives me an error in the build of rotors_gazebo_plugins

RugvedKatole, 2024-04-04:
Hi, I am facing the same error. I ran all the commands you provided here. 
I have attached the output given by the catkin build command.
[errors.txt](https://github.com/gsilano/CrazyS/files/14875405/errors.txt)

gsilano, 2024-04-05:
Hi there! Have you tried following the commands mentioned above?

gsilano, 2024-04-14:
@RugvedKatole , read #40 and let me know if this solves the issue.

Faivex, 2024-06-24:
Hi all,
I had almost the same problem.
I've been trying to build CrazyS on Windows Subsystem for Linux (WSL), using ```Ubuntu 20.04```, ROS 1, and ```Gazebo 11```.
Every time I tried to build ```rotors_gazebo``` and ```rotors_simulator``` an error was raised.
As a result, when I launched the hovering example, the drone didn't fly.
After testing many possible solutions, I discovered it is because the swap is 1GB on my WSL (you can find it by ```swapon --show```), which is insufficient. 
I simply increased it to 4GB (see [here](https://www.youtube.com/watch?v=Tu95sdnALJk)) and the problem was solved.
Hope this helps.

kashifzr, 2024-07-30:
@gsilano i need to know how to use plugins define like librotors_gazebo_lidar_plugins.so define in rotors_gazebo_plugins. I am using lidar in component snippet where different plugins with cpp code are given. When i used plugins given by gazebo like libgazebo_ros_laser.so it give me scan values and it does work as expected but I want to used some customized plugins in my codes.
<pre><font color="#06989A"><b>libLiftDragPlugin.so</b></font>
<font color="#06989A"><b>libroll_pitch_yawrate_thrust_controller.so</b></font>
<font color="#06989A"><b>librotors_gazebo_bag_plugin.so</b></font>
<font color="#06989A"><b>librotors_gazebo_controller_interface.so</b></font>
<font color="#06989A"><b>librotors_gazebo_fw_dynamics_plugin.so</b></font>
<font color="#06989A"><b>librotors_gazebo_gps_plugin.so</b></font>
<font color="#06989A"><b>librotors_gazebo_imu_plugin.so</b></font>
<font color="#06989A"><b>librotors_gazebo_lidar_plugin.so</b></font>
<font color="#06989A"><b>librotors_gazebo_magnetometer_plugin.so</b></font>
<font color="#06989A"><b>librotors_gazebo_motor_model.so</b></font>
<font color="#06989A"><b>librotors_gazebo_multirotor_base_plugin.so</b></font>
<font color="#06989A"><b>librotors_gazebo_noisydepth_plugin.so</b></font>
<font color="#06989A"><b>librotors_gazebo_odometry_plugin.so</b></font>
<font color="#06989A"><b>librotors_gazebo_pressure_plugin.so</b></font>
<font color="#06989A"><b>librotors_gazebo_ros_interface_plugin.so</b></font>
<font color="#06989A"><b>librotors_gazebo_velodyne_plugin.so</b></font>
<font color="#06989A"><b>librotors_gazebo_wind_plugin.so</b></font>
<font color="#3465A4"><b>pkgconfig</b></font>
<font color="#3465A4"><b>python3</b></font>
<font color="#3465A4"><b>rotors_control</b></font>
<font color="#3465A4"><b>rotors_evaluation</b></font>
<font color="#3465A4"><b>rotors_gazebo</b></font>
<font color="#3465A4"><b>rotors_joy_interface</b></font>
</pre>
in component_snippet all these plugins are being referred but librotor_gazebo_lidar_plugin is not used but given in rotors_gazebo_plugins. Hope this make sense

gsilano, 2024-07-31:
Hi! I think you should use the [issue tracker of the RotorS package](https://github.com/ethz-asl/rotors_simulator) for this type of question.

IngCristianVC, 2024-08-30:
> Hi all, I had almost the same problem. I've been trying to build CrazyS on Windows Subsystem for Linux (WSL), using `Ubuntu 20.04`, ROS Noetic, and `Gazebo 11`. Every time I tried to build `rotors_gazebo` and `rotors_simulator` an error was raised. As a result, when I launched the hovering example, the drone didn't fly. After testing many possible solutions, I discovered it is because the swap is 1GB on my WSL (you can find it by `swapon --show`), which was insufficient. I simply increased it to 4GB (see [here](https://www.youtube.com/watch?v=Tu95sdnALJk)) and the problem was solved. Hope this helps.

hey, the video says that is private, i had this problem with the catkin build

gsilano, 2026-10-13:
Closed for inactivity

closed 2026-10-13 by gsilano