gsilano/CrazyS #109: Extending the model

vbalaji21, 2024-04-12:
I am extending the multi-rotor model by adding more links. I am adding the Gazebo Plugins and I am unable to see the links when spawned and launched. Can you please let me know what is the right way to add plugins in this framework.

The way I did is adding the header and cpp files for thaose links similar to other links and adding those files as plugins in the model description (SDF).

gsilano, 2024-04-14:
Hi there! Could you provide more specific details? For example, adding a sketch of the file you've modified and explaining how you're using it would be helpful. Additionally, providing information about the operating system and the ROS release you're working with would greatly assist in understanding the issue. Thanks!

gsilano, 2026-10-13:
Closed for inactivity

closed 2026-10-13 by gsilano