gsilano/CrazyS #110: overactuated hexacopter to quardcopter

MayurRabadiya, 2024-04-29:
Hello There,
I am building an overactuated quadcopter with the help of a hexacopter available in omav in this repo.

I am facing an issue with the publisher:

rostopic pub /omav/command/motor_speed mav_msgs/Actuators '{angular_velocities: [500, 500, 500, 500, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]}

Here are 18 variables available. 6 upper rotors + 6 lower rotors + 6 tilting arm. 

in  my case only 8 variable, 4 rotor and 4 tilting arm are there.   

I am trying to update this 18 variable to 8 variable. But I am not able to do. 

Can someone help me with how I can update the variables to 8? 

And One more time, 
If I remove the hovering_example node running from the launch file, I am not seeing any topics in the rostopic list. 

How can I get all the topics without hovering_example node?? Is there any documentation or source available for that?

Thank you very much in advance. 

Kind regards,
Mayur

gsilano, 2024-05-05:
Hi there! Could you please specify the file you are referring to, as well as the version of ROS and the specific version of CrazyS that you are using? This information will help me provide you with more accurate assistance. Thanks!

MayurRabadiya, 2024-05-05:
Hello @gsilano, Thank you for the reply.

I am using ROS noetic, and I have cloned the ros/noetic branch.

I am working with omav.xacro, omav_base.xacro and underactuated.xacro. I have copied these files from original rotors_simulator  repo.

I want to make a tilted-rotor drone.

gsilano, 2024-05-14:
Hi! I was searching for the `omav.xacro` and `omav_base.xacro` files in the `dev/ros-noetic` branch (https://github.com/gsilano/CrazyS/tree/dev/ros-noetic/rotors_description/urdf), but unfortunately, I couldn't locate them. Could you please provide the correct link?

Thanks!

MayurRabadiya, 2024-05-14:
[https://github.com/ethz-asl/rotors_simulator/tree/master/rotors_description/urdf](url)
Hii,
Here, are the files. I am using this files.

gsilano, 2024-05-19:
Okay, now I see. Since this multi-rotor was never tested with CrazyS, I'd start with a simple example. I'd create an empty world by modifying the word file [crazyflie2_without_controller.launch](https://github.com/gsilano/CrazyS/blob/dev/ros-melodic/rotors_gazebo/launch/crazyflie2_without_controller.launch). Then, I'll try manually publishing the rotor velocities to the vehicle. If that works, you are on the right track.

Next, you'll need to create your own controller or import one from the RotorS package. Perhaps, they designed a proper version of Lee's controller or any other. Finally, you need to create your own trajectory generation file, as it strictly depends on the controller you are using. For instance, it might require not only position references, but also velocity, acceleration, and jerk.

I hope this helps.

MayurRabadiya, 2024-05-23:
Thank you very much!! @gsilano

closed 2024-05-23 by MayurRabadiya