gsilano/CrazyS #111: External MPC controller not working properly

rakesh55ydv, 2024-11-19:
Hello gsilano,
Thanks for you work in creating this wonderful models to work in SITL. I am facing several issues:

1) I created an MPC controller in matlab to track a helical trajectory and it is working fine. However, when I publish the omega of the rotors in the topic /crazyflie2/gazebo/command/motor_speed , it is not behaving as it should be and model becomes unstable. I played with the rates of the controllers. However, it did not have any affect in the simulation. 
2) Therefore, I decided to use inbuilt position_controller_node.cpp by passing the states coming from my MPC controller in the topic called /crazyflie2/command/trajectory. The position controller node keeps the crazyflie stable, but it is very slow and could not track the trajectory I am publishing. I have kept the trajectory publishing rate fairly low. 

Are there any suggestions on how can I make my external controller work ? How to use the inbuilt controllers in this scenario ? 

Thanks,
rakesh55ydv

gsilano, 2024-11-19:
Hi @rakesh55ydv! I'm glad you found my work interesting. 

Have you tried following the [guide in the Wiki](https://github.com/gsilano/CrazyS/wiki/Interfacing-CrazyS-through-MATLAB) to connect MATLAB with Gazebo? This guide explains how to establish message exchange between the two. While more advanced solutions for synchronization exist today, the method described in the guide should suffice for initial testing.

If you continue to experience instability issues, the problem might lie in the parameters used for the MPC design not matching those in Gazebo. In this case, I recommend reviewing the following files:  
- [File one](https://github.com/gsilano/CrazyS/blob/dev/gazebo9/rotors_description/urdf/crazyflie2.xacro): Based on my own identification process.  
- [File two](https://github.com/gsilano/CrazyS/blob/dev/gazebo9/rotors_description/urdf/crazyflie2_forster.xacro): Derived from an [ETH thesis](https://www.research-collection.ethz.ch/handle/20.500.11850/214143).

The key point is to ensure consistency between the parameters used in your MPC and Gazebo simulations. If needed, you can modify these files to create a Crazyflie setup tailored to your needs.

Regarding the launch file, if it isn't included in the Wiki guide, I suggest creating an empty Gazebo world containing only the Crazyflie with the appropriate configuration settings. An example is provided in this [launch file](https://github.com/gsilano/CrazyS/blob/dev/gazebo9/rotors_gazebo/launch/crazyflie2_without_controller.launch).

Lastly, when setting up your environment, ensure that the ROS version matches your system's configuration (e.g., Kinetic or Melodic for Ubuntu 18.04, or Noetic for Ubuntu 20.04). You can find a complete list of supported ROS releases in the README.md file.

I hope this helps

gsilano, 2026-10-13:
Closed for inactivity

closed 2026-10-13 by gsilano