gsilano/CrazyS #113: Regarding Stable Rotor Speed Control

rodhakate, 2025-04-04:
Hello, 

So I have been using CrazyS to train a RL controller.

Instead of learning for high-level control (Position control,  R-P-Y-Thrust control) I want to learn directly the rotor speeds to control the quadcopter (I am using the Asctec HUmmingbird). 

So during the training I am running several steps per episode (~500 steps), i.e. to say I am applying the RL generated actions (rotor speeds) to the quadcopter as, 

[This is just ax example snippet]

```
self.rotor_msg = Actuators()
self.rotor_cmd_buffer = [0.0, 0.0, 0.0, 0.0]

def start_rotor_publisher(self, rate_hz=200):
        if not self.rotor_publisher_started:
            rospy.Timer(rospy.Duration(1.0 / rate_hz), self._rotor_timer_callback)
            self.rotor_publisher_started = True
            rospy.loginfo(f"Rotor command publisher started at {rate_hz} Hz.")

def _rotor_timer_callback(self, event):
        self.rotor_msg.angular_velocities = self.rotor_cmd_buffer
        self.rotor_pub.publish(self.rotor_msg)

def set_rotor_command(self, rotor_speeds):
        self.rotor_cmd_buffer = rotor_speeds
        rospy.sleep(0.05)
```
Currently I am training the quadcopter just to hover. So I am giving same rotor speeds to all 4 rotors. 
So the quadcopter should just move in the z direction.

However after certain number of steps the quadcopter goes out of control and just crashes somewhere. I know from the RotorS description, where they mentioned,

>  **The multicopter is unstable though, since there is no controller running, if you just set the motor speeds.**

So for the hovering task, I can just put some bounds on the x and y and reset the environment if the quadcopter goes out of bounds. 
But In order to train for target or trajectory tracking, the quadcopter going unstable will pose critical problems in training.

Is there a way to use rotor speed control [Actuators()] in a proper way.

Thanks in advance.
Kind Regards,
Rohit

gsilano, 2025-04-07:
Hi Rohit,

Your project sounds really interesting, and you're pushing into a more challenging area by trying to control the quadcopter directly through rotor speeds. As you've noticed, this approach brings a lot of instability, especially since you're bypassing the usual flight controller that handles stabilization.

`RotorS` does mention that the multicopter is unstable without a controller; this is expected behavior. Quadcopters aren't passively stable, so even something that seems simple, like hovering, requires constant motor adjustments. When you directly set motor speeds (e.g., equally on all four rotors), any small disturbance or modeling error can quickly grow and send the drone off course.

Here are a few suggestions that might help:

1.  **Add a Low-Level Stabilization Layer.** Even if you're using RL, it might help to introduce a basic PID controller that runs underneath the RL policy. This kind of hybrid setup is pretty common in robotics (https://arxiv.org/abs/2006.05768, https://doi.org/10.1109/LRA.2017.2720851): the RL policy learns higher-level commands, and a traditional controller handles the fine-grained stabilization. That way, you avoid crashes while still training the policy effectively.

2. **Use Environment Constraints During Training.** You already mentioned bounding the `x` and `y` axes and resetting the environment if the drone drifts too far: that’s a good idea. It's a practical way to prevent the simulation from going unstable too often. For the early stages of training, it can really help keep the learning process on track.

3. ** Start Simple, Then Scale Up.** It might be helpful to first train the agent on very simple tasks, like just maintaining a stable altitude (`z-axis` only), before attempting trajectory following. Once it masters that, you can gradually introduce more complex behaviors. This curriculum-style learning can make training much smoother.

4. **Carefully Tune Your Training Parameters.** Sometimes instability can come from the RL side too: overly aggressive learning rates, poorly scaled rewards, or action spaces that are too wide. It's worth taking time to tune those elements carefully. Also, consider action smoothing if your RL outputs are jumping too much between steps.

In short, direct rotor control is possible, but some form of stabilizing structure, either in the environment or the controller, usually helps a lot. Good luck with your training!

gsilano, 2026-10-13:
Closed for inactivity

closed 2026-10-13 by gsilano